elechouse/VoiceRecognitionV3 #3: when I execute a command with the V3 I keep getting an error message

newtotech, 2016-05-23:
Every time I try to execute a command via serial com I keep getting an error message like "Train with signature failed or timeout." or more like "(command) failed or timeout" but the lights indicate that it has worked. It also will not pull the any signatures or check records and when I input the settings command I get the "settings failed or timeout." response.

am I doing something wrong?

Oh, I'm using the V3 module with my Arduino 101 board. the txd and rxd pins are connected to my tx and rx digital input pins 0 and 1.

wilson-elechouse, 2016-05-23:
You seems to connect it to the wrong pins. The VR3 board should be connected to softserial pins                                                                                                                                                                                                                                                                                                                                         Wilson ShenBusiness ManagerFrom BlackBerry smartphone.                                                                                                                                                                                                                From: newtotechSent: Monday, May 23, 2016 08:49To: elechouse/VoiceRecognitionV3Reply To: elechouse/VoiceRecognitionV3Subject: [elechouse/VoiceRecognitionV3] when I execute a command with the V3 I keep getting an error message (#3)Every time I try to execute a command via serial com I keep getting an error message like "Train with signature failed or timeout." or more like "(command) failed or timeout" but the lights indicate that it has worked. It also will not pull the any signatures or check records and when I input the settings command I get the "settings failed or timeout." response.

am I doing something wrong?

Oh, I'm using the V3 module with my Arduino 101 board. the txd and rxd pins are connected to my tx and rx digital input pins 0 and 1.

newtotech, 2016-05-31:
Can you help me identify the "soft serial" pins on the Arduino 101?

I have tried everything I know of and I cannot get it to work.

wilson-elechouse, 2016-06-01:
please refer to this video
https://vimeo.com/94925137

newtotech, 2017-04-23:
After my last comment about purchasing a new V3 board, I gave up and just put the board away. I pulled the board back out about a week ago and have been looking into the issues I am having. It appears the indicator light just continues to stay on. When I input "settings" or "vr" I get the message "check system settings error or timeout" but when I input "sigtrain 0 ON" the light starts blinking and switches from red to yellow like it's working. This tells me that I have everything hooked up correctly and the V3 board is receiving my instructions. However, the serial monitor prints "Train with signature failed or timeout." before the lights even stop blinking. 

any idea what could cause this?

wilson-elechouse, 2017-04-24:
We suggest you try the following way to test it.
https://vimeo.com/94926818
You could use Arduino as USB-TTL

Misiukas1966, 2017-09-26:
Good afternoon, Wilson!
My voice recognition board has stopped working the lights blink but it does not receive any commands, it has some idea of what could have happened because it worked very well.
Using vr_sample_train I always get the return
Check recognizer failed or timeout.

Thank you
Antonio
[image]

wilson-elechouse, 2017-09-27:
Make sure the wire connection is correct. It seems to be the communication
problem.

Misiukas1966, 2017-09-27:
Thanks for the quick return Wilson!
The conccoes are certain this happened when recording a new command maybe the sequence of lights tell you what can be. Is there any way to reset the board?
Follow the link with the video of the board;

https://www.youtube.com/watch?v=XcwN62FpGvc&feature=youtu.be

Antonio

Sorry for my bad English :(

wilson-elechouse, 2017-09-29:
Hi,
if the module could not receive any data, this command occurs.
Possible reasons:
1. Wire connection problem
2. UART interface damage. if you have ever connected TX/RX to GND/VCC, the
pins might be burned.

Try to bridge TX and RX, then power it up. It is hardware reset method.

Misiukas1966, 2017-09-29:
Thank you again!
I believe the pins are burned. I'll ask for a new board on your site soon probably to simpleVR.
Thank you very much for your attention.
Antonio

rajeshraj1998, 2018-04-25:
same problem in mine also, how to hardware reset the v3 module

lfg6000, 2018-11-28:
Not all inputs will work for SoftwareSerial.
I learned this the hard way.
Look at digitalPinToPCICR() in pins_Arduino.h to see what pins i/o will work.

SoftwareSerial on the AVR  Adafruit ItsyBitsy 32u4	
- rx/tx will not work on pins 0,1,2,3,5...learn this the hard way
- SoftwareSerial is a simulated UART...it uses regular i/0 pins to simulate a dedicated UART
- must use pins 8,9,10,11,14,15,16,17,A8,A9,A10 for Rx and Tx
  digitalPinToPCICR() is used in SoftwareSerial::begin 
  #define digitalPinToPCICR(p)    ((((p) >= 8 && (p) <= 11) || ((p) >= 14 && (p) <= 17) || ((p) >= A8 && (p) <= A10)) ? (&PCICR) : ((uint8_t *)0))
  located in ....\adafruit\hardware\avr\1.4.13\variants\itsybitsy32u4\pins_arduino.h

ulum34, 2020-02-27:
Hi, i have a problem in my voice recognition board
[image]

why does the output keep coming out continuously even though I haven't said a word ?
thank you

vaibhavgosavi10, 2020-03-16:
Can I use vr 3 library for vr1

mathewcigi1997, 2021-12-06:
This code wont work on higher version of Arduino IDE. Please use **Arduino v1.0 or v1.0.6**
Also change "_myVR.begin(9600);_ " to "_myVR.begin(19200);_ " in the vr_sample_train
I hope this work for all ^_^

Don't forget to look that your connections are proper.
5v == 5v
GND == GND
RXD == pin 3
TXD == pin 2

I used Arduino UNO. I think this will work for every Arduino board.

IslamEmran, 2022-03-02:
You guys should connect the pin TXD from VR unit With any digital pin, but it must be taken into account that it is not (PWM)
As for the pin RXD, it must be connected to any digital pin, but provided that it is (PWM)

For example, if you are using an Arduino Mega, you can take the pins:

Arduino Mega                 VR

GND-------------------> GND

VCC------------------->5V

50------------------->TXD

3------------------->RXD

RealTech360, 2022-10-04:
[image]
I have face 2 problem

RealTech360, 2022-10-04:
I have face 2 problem1st problem train voice not working, 2nd 
[image]

BilalKhalid-22, 2022-12-27:
> Every time I try to execute a command via serial com I keep getting an error message like "Train with signature failed or timeout." or more like "(command) failed or timeout" but the lights indicate [...]

I was doing the same thing after spending a lot of time i came to know that TXD pin should be connected with RX pin of Arduino and RXD pin should be connected with TX pin of Arduino

ETH65, 2023-02-17:
Hello, 
I am using V3.1 with ESP32 Feather: The VR3 responds correctly to the 'help', 'train', 'getsig', 'record', 'clear', and 'vr' commands. 
It does not, however LOAD any records to the Recognizer:  the load command returns “Load failed or timeout.”  'vr' command returns no records loaded.
Any ideas, please?

Setup: 
Adafruit ESP32 Feather (“Huzzah32”)
Elechouse VR3.1
External power supply for 5V  (resistor bridge for VR3 TX -> RX on the ESP32)

Software: 
Github: elechouse/VoiceRecognitionV3 by Wilson-elechouse, dated Jan11, 2017
Installed modified VoiceRecognitionV3.h and VoiceRecognitionV3.cpp from here: 
[https://blog.frankvh.com/2022/02/21/esp32-support-for-elechouse-voice-recognition-module-v3/]

Operation: 
I am running the vr_sample_train program. 
It compiles and loads without errors. 
Communication with the VR3 is possible, as commands with the VR module and Recognizer are being answered.

Error condition:  
the load command returns “Load failed or timeout.” No record is loaded onto the Recognizer: 'vr' command returns no records loaded.

lllewsor, 2023-11-01:
Hello guys !
I had a similar problem with an Arduino Mega 2560. The problem was with the wiring. I finally connected TX to pin 50 and RX to pin 3 and... bingo !

khaled312001, 2024-01-09:
Did somebody find any solution?

ETH65, 2024-01-10:
Hello, I was able to resolve the problem that I reported in this chat on Feb 16th, 2023. 
Here's what I found (2 parts): 
(1) I did a UART trace and found that the load instruction seemed incomplete. So I amended the file VoiceRecognitionV3.cpp, adding two lines 1203 and 1204. The modified file is here attached as *.zip file.
(2) the ESP32 operates with 3.3V while the VR3 operates with 5V. The 3.3V is less than 70% of required CMOS voltage therefore the ESP TX messages are not recognized reliably by the VR3. I used a level shifter SN74AHCT125N to adjust voltage levels.
With those two modifications the VR3 works very reliably with the ESR32 in my application. 
I hope this helps.  
[VoiceRecognitionV3.zip](https://github.com/elechouse/VoiceRecognitionV3/files/13881471/VoiceRecognitionV3.zip)

khaled312001, 2024-01-10:
thank

> Hello, I was able to resolve the problem that I reported in this chat on Feb 16th, 2023. Here's what I found (2 parts): (1) I did a UART trace and found that the load instruction seemed incomplete. [...]

thank you very much

khaled312001, 2024-01-10:
> Hello, I was able to resolve the problem that I reported in this chat on Feb 16th, 2023. Here's what I found (2 parts): (1) I did a UART trace and found that the load instruction seemed incomplete. [...]

how run this code with arduino uno in arduino ide please

ETH65, 2024-01-10:
Hi, I don't know exactly how to do this in your case and set-up.
This is what I would try:

  *    unzip the file that was included in my response
  *   Copy the file into the VR3 library. You may already have a file with the same name in the VR3 library: replace that file with the new one that you just unzipped.
  *   Make sure you have the Include command in your Arduino code and that the settings point to the correct library

good luck!

Thanks and best regards
Eckart

khaled312001, 2024-01-10:
can you help me to fix loading error after train module didn't load the voice , i use arduino uno and vr V3.1
[image]
[VoiceRecognitionV3.zip](https://github.com/elechouse/VoiceRecognitionV3/files/13888731/VoiceRecognitionV3.zip)

ETH65, 2024-01-12:
Hi,
The library file I embedded in the chat is customized for ESP32. I am not sure if that library file works for the UNO.
I suggest that you download the library of the same name from GitHub and then add the two lines that I modified in "my" version.

Thanks and best regards
Eckart

image.png (view on web)<https://github.com/elechouse/VoiceRecognitionV3/assets/58705759/8a346a79-ec2d-4c1e-ab86-62394915cee9>

khaled312001, 2024-01-12:
> Hi, The library file I embedded in the chat is customized for ESP32. I am not sure if that library file works for the UNO. I suggest that you download the library of the same name from GitHub and then add the two lines that I modified in "my" version. Thanks and best regards Eckart
> […](#)

[image]
Are these the two lines should I add?

ETH65, 2024-01-12:
Yes, those two lines.

Thanks and best regards
Eckart

Screenshot_2024-01-12-23-41-03-574_com.foxdebug.acode.png (view on web)<https://github.com/elechouse/VoiceRecognitionV3/assets/58705759/ada29354-f2f5-4252-8c82-39244ec06a5e>

peteleba, 2024-12-16:
I am encountering same issue, I used the VR3 module on saturday and it worked perfectly, tried again and it now gives train timeout. i used both the arduino smd and arduino uno board. what could be the issue?