elechouse/VoiceRecognitionV3 #4: Check system settings error or timeout

nobodykid, 2016-08-15:
I got similar error with #3 , but mine when I tried to send `settings` command to the module

The message written can be seen in screenshot provided

[image]

I developed the code in Windows, but tried to send this command from my Ubuntu PC. The code worked well in Windows. 

Could it be a problem on me changing PC or the problem in module?

wilson-elechouse, 2016-08-17:
if the code works well Windows, the module should have no problem.
Please check the wire connection to make sure it is right.

nobodykid, 2016-08-18:
I see. Thanks for the reply

I also noticed on your documents about certain VR3 feature : 

`Recognize maximum 7 voice commands at same time`

Does it mean that if I give several commands at the same time, the module will be able to recognize each of them up to 7 different commands?

wilson-elechouse, 2016-08-19:
yes, correct

nobodykid, 2016-08-21:
Thank you.

I also looking for some hardware specifications, like datasheets. Do you have that available for this device?

safari2, 2017-03-10:
Please add a comment, that if you use MEGA2560, you should change ping to 10,11 (2,3 don't maintain SoftwareSerial MEGA 2560);
// 2:RX 3:TX, you can choose your favorite pins 
its not correct phrase, because not all pins on AVR processors maintains interrupts, needs for RX!!!

trychaga, 2017-04-10:
Hie
I am using Funduino Mega 2560(Atmega 2560) and VR3 it seems when i do the connections as outlined at the start of the VR3 tutorial and change serial pins to 0&1 on the serial monitor i get the error : **Check system settings or timeout**
also on my recognizer i only see one led light for power is coming on, is it a hardware problem since i soldered the vr3 pins on my own or its a software problem?
if i try a different command it says Command unrecognized
May you help

wilson-elechouse, 2017-04-10:
if you use MEGA2560, you should change ping to 10,11

trychaga, 2017-04-22:
> [image: Boxbe] <https://www.boxbe.com/overview> This message is eligible
> for Automatic Cleanup! (notifications@github.com) Add cleanup rule
> <https://www.boxbe.com/popup?url=https%3A%2F%2Fwww.boxbe.com%2Fcleanup%3Fkey%3DyqsQeu9BiEg49KOBzeuDqSmrsOlyloWvWExKdSLw8W8%253D%26token%3DyZonL8JuiJZhJGQ4Wlm4NuCYvheLq16ox1UEtbaa94WMZDVYBnZvHTZ7mVQ5131OxAgMq3XrCMX90tRg49ERafTe8XzW8RvzpKE6ybIRUVpnf5TuuriLtPUpq%252F2XfBdcSq%252Flq11tmsSde1IvBz%252Ft8g%253D%253D&tc_serial=29738523661&tc_rand=673730014&utm_source=stf&utm_medium=email&utm_campaign=ANNO_CLEANUP_ADD&utm_content=001>
> | More info
> <http://blog.boxbe.com/general/boxbe-automatic-cleanup?tc_serial=29738523661&tc_rand=673730014&utm_source=stf&utm_medium=email&utm_campaign=ANNO_CLEANUP_ADD&utm_content=001>
>
> if you use MEGA2560, you should change ping to 10,11

Dear Wilson
Thank you very much i managed to configure the module after changing the
ping to (10,11) and a managed to record my voice as well.
However i still have a small challenge I'm designing a security system
using this system and i like the response rate and recognition rate of this
module but two in three test users seem to share the same voice signature
which i do not like because there wont b any security at all.
How can I go about this? Is it that I have to record and train using high
pitch always or is it about how i train it. Where i am missing it is how
many times should i train my voice to this module before it begins to
recognize me and only me. Please help.
My other question and comeback is i wish to provide an audio feedback or
acknowledgement when the signature s correct and how can i achieve this
using the VR3.1 module if i do not want to interface the Funduino Mega 2560
with an SD Can i access the pre-recorded voice recordings from the VR3
group and achieve playback? If so how can i achieve this?
Regards

htorus, 2017-08-14:
Hi! I'm using Arduino Mega 2560 R3. The cable connection is correct but I'm facing this problem. I can not record sound. How can I solve this problem?

shinny1711, 2019-02-10:
hi,
i connected voice module v3 with arduino  mega such as 
vcc-5v
gnd-gnd
tx-10
rx-11
when i connected usb cable to arduino then light in the arduino is not glowing.
but when i disconnect either vcc or gnd pin from voice module then the light is glowing but the light in the voice module is not glowing in any way. why???
does my my voice module v3 is working or not???
how to know whether it is working or not?

mzyada, 2019-09-20:
hi everyone, when I try my vr3 it stopped working! I don't know what happened, anyone has an idea or solution?? plz help me.

hamzi7860, 2020-03-16:
> Please add a comment, that if you use MEGA2560, you should change ping to 10,11 (2,3 don't maintain SoftwareSerial MEGA 2560);
> // 2:RX 3:TX, you can choose your favorite pins
> its not correct phrase, because not all pins on AVR processors maintains interrupts, needs for RX!!!

I'm Using AT-Mega2560 and pins 10,11 are working for me, Thank You

hamzi7860, 2020-03-16:
> […](#)

Yes you are right , Thanks it worked for me :)

AKQ8, 2020-03-19:
hi,
im using Arduino Uno and Geeetech Voice Recognition, and i use the "train" sample from the libraray and try to type settings or any other command i get the following error.
[image]

hamzi7860, 2020-03-19:
hey! can you please mention about pins you are using for TXD and RXD on arduino uno. You are getting this error because your pin configuration is not correct.

AKQ8, 2020-03-19:
> hey! can you please mention about pins you are using for TXD and RXD on arduino uno. You are getting this error because your pin configuration is not correct.

TXD -> 2
RXD -> 3
GND -> GND
VCC -> 5V

hamzi7860, 2020-03-19:
use this pin configuration without doing any change in code. just change wires of module. your module will start working.
TXD -> 3
RXD -> 2
GND -> GND
VCC -> 5V

AKQ8, 2020-03-19:
still getting error, with the changes you provided above.

hamzi7860, 2020-03-19:
then please attach a screenshot of code (which lines include pins configuration)

AKQ8, 2020-03-19:
im using the example "train" from the library

hamzi7860, 2020-03-19:
check your connections and pin configuration again and play with RXD and TXD pins by replacing their positionsposition with 2 or 3. It will gona work.

AKQ8, 2020-03-19:
still getting same error, but when i configure the following, commands dont respond to the monitor
TXD->0 
RXD->1

hamzi7860, 2020-03-19:
if you are using arduino uno. use only 2 and 3 pins of arduino for TXD and RXD in code but replace their positions on hardware only.

AKQ8, 2020-03-19:
still getting the error even with pins 2 and 3.

AKQ8, 2020-03-19:
my Arduino Uno Info:
BN: Arduino Uno
VID: 0x2341
PID: 0x0043
SN: 857333239393516141B2

AKQ8, 2020-03-20:
if anyone got the fix for this issue please help out :)

Charisavena97, 2020-06-06:
@AKQ8 I have the same problem, did anyone find the solution?

AKQ8, 2020-06-07:
unfortunately, still issue occurs and no has given/found a solution

@Charisavena97

Devi1136, 2021-05-03:
hii, im using Ardduino Nano board. Facing the same problem. Every mentioned command in the serial monitor is failed or timed out. I tried both the connections 1) Rx : 2 , TX : 3 ( Recommended by the Sampletrain program by VR3 lib) ...2) RX : 2 , TX : 1 (Built in Pins of arduino nano ) . Still the commands fails...... And one more thing the led on voice recognition module is always on....not blinking as i see in the tutorials in youtube about VR3. Please Help...

taf2, 2022-01-08:
@Devi1136 i found that for arduino uno board at least I had to reverse in the code 2 and 3 until it works.  So what I did is wire 2 and 3 to the board's RX and TX.  Then in software I reversed the 

```
VR myVR(3,2);    // 2:RX 3:TX, you can choose your favourite pins.
```

and now it works great.

Aswini2414, 2022-03-02:
@taf2 I checked in the way u told above, but still i am get the error

IslamEmran, 2022-03-05:
@Aswini2414 

https://github.com/elechouse/VoiceRecognitionV3/issues/3#issuecomment-1056161112

pumpkinandknife, 2023-01-05:
Did u try uploading the code? That worked for us and we found that a lot of ppl we know didnt do this. :3

Rickum1, 2023-04-11:
can yall suck my balls_

swseansw, 2023-12-09:
My experience for reference: Replace that W25Q16JV chip. Can purchase on eBay or Amazon shop.  I guess there are more than one producers, as these Chinese producers always copy each other's products.