elechouse/VoiceRecognitionV3 #9: can't train Voice Recognition Module VR3

snz007, 2017-10-12:
i got problem with my voice recognition module like that 

[image]

baud rate and pulse width  unknown cause i cant train this module. 
how i solve this problem?

hbadotti, 2017-12-27:
Post your code, your board, your pinage.....

VirtualEnder, 2023-10-12:
I'm running into the same issue. This using the sample code vr_sample_train.ino from https://github.com/elechouse/VoiceRecognitionV3/blob/master/examples/vr_sample_train/vr_sample_train.ino

Arduino Uno R3 wired as shown in the sample code.  When I type settings in the serial monitor I get: 

```
--------------------------------------------------------------------------------
settings
--------------------------------------------------------------------------------
Baud rate: UNKONOWN
Output IO Mode: UNKONOWN
Pulse Width: 20ms
Auto Load: enable
Group control by external IO: UNKNOWN STATUS
``` 

I can run through sigtrain and train, and it says 

```
Output IO Mode: UNKONOWN
Pulse Width: 20ms
Auto Load: enable
Group control by external IO: UNKNOWN STATUS
--------------------------------------------------------------------------------
sigtrain 0 On
--------------------------------------------------------------------------------
Record:	0	Speak now
Record:	0	Speak again
Record:	0	Cann't matched
Record:	0	Speak now
Record:	0	Speak again
Record:	0	Success
Success: 1
Record 0	Trained
SIG: On
--------------------------------------------------------------------------------
``` 

But when I check the record it says:

```
--------------------------------------------------------------------------------
record 0
--------------------------------------------------------------------------------
Check 1 records.
0 record trained.
0	-->	Untrained
--------------------------------------------------------------------------------
``` 

And load fails as well.  It appears that even though it says the training was successful, it isn't stored in the module.  Any help would be appreciated.

khaled312001, 2024-01-09:
are you solve this problem?

khaled312001, 2024-01-09:
> I'm running into the same issue. This using the sample code vr_sample_train.ino from https://github.com/elechouse/VoiceRecognitionV3/blob/master/examples/vr_sample_train/vr_sample_train.ino
>
> Arduino [...]

are you solve this problem?

paisp, 2024-02-02:
I have the same problem related above. With arduino uno and  mega 2560. My project stoped. Can we have an answer very quickly. As we have some boards that work, can you pleas tell where can i by right now. Thanks in advanced. @paisp