elechouse/VoiceRecognitionV3 #13: ESP32 compatibility

arduibag, 2018-12-16:
Is it possible to have an ESP32 compatible library ?
It would be awsome !

asianplanet, 2021-01-15:
IDE doesn't even compile for ESP32

asianplanet, 2021-01-17:
no support for ESP32 so far

ilhmfahmi, 2021-03-18:
> no support for ESP32 so far
yeahh me too, so how to use with esp32?

arduibag, 2021-03-19:
I found the solution. 

The problem is about the voltage difference between the Voice Recognition module (5v) and ESP32 (3.3v).

So they can't communicate together ! haha

You need to use this:
level shifter
https://www.sparkfun.com/categories/361

with this:
DC converter
https://www.amazon.fr/ANGEEK-0-9-5V-Step-Up-Voltage-Converter/dp/B07ZDJPMPJ

... between your ESP32 and your VoiceRecognition module and then they will be able to communicate together.
So no change on the software but just some electric modification.... :-)

arduibag, 2021-03-19:
And don't forget to use only dedicated RXD/TXD pin of the ESP32 too (depends on your ESP32 module).

ilhmfahmi, 2021-03-31:
> And don't forget to use only dedicated RXD/TXD pin of the ESP32 too (depends on your ESP32 module).

[image]

esp32 have 5v voltage, i already connect the module but its still the same.

VictorNrt, 2021-09-21:
Hi, I'm trying to use this module with ESP32 but the sketch doesn't upload when I include VR3 library... 
Could somebody tell me how to use the module with an ESP32 ? 

Thank you for your helop 
Victor from France

taf2, 2022-01-08:
I think if you reverse what is documented at tx and rx it might work for you...

javiersotogarciadevillegas, 2023-02-05:
> I found the solution.
>
> The problem is about the voltage difference between the Voice Recognition module (5v) and ESP32 (3.3v).
>
> So they can't communicate together ! haha
>
> You need to use this: level [...]

ese modulo es para alimentar al elechouse  , osea sacarle los 3.3 voltios al esp32 y convertirlos en 5v para el elechouse?

zaid-alrowaili, 2023-09-22:
I have a problem saving the audio 
[image]

sysshad, 2024-03-30:
Hello all, i use Wemos lolin ESP32 S2 mini, and i can share my changes that it took me to train and use this little module.
The VoiceRegonition module and the ESP32 both are powered by 5V. But the Serial pins on the ESP32 is not 5V, they are 3.3V

1)
So firstly you must put a level shifter in between the serial connections between the devices. This can be done with either resistors, or better up MOSFET transistors. There exists alot of these to buy.

2)
The VoiceRecognitionV3.h file references the library avr/pgmspace.h and this does not exist/work for ESP. So remove/comment this line. This library isnt even used by voicerecognition code...

3) If the MCU you have uses Serial through the USB protocol then the device needs this code to wait for your PC to establish connection. As was the case for my Wemos lolin S2 mini...
while (!Serial) delay(100);
Serial.begin(115200);

4) I would also change the VoiceRecognitionV3.h/cpp files to use the HardwareSerial instead of SoftwareSerial. Most ESP32 devices have 2+ Serials.