elechouse/VoiceRecognitionV3 #14: voice recognition not blinking and can't load the train

sayakingXD, 2019-06-15:
my voice recognition v3 not blinking and why ican't load the train
[image]

Core-drummer, 2019-10-11:
I have the same problem. 
`Elechouse Voice Recognition V3 Module "train" sample.
--------------------------------------------------------------------------------
Usage:
--------------------------------------------------------------------------------
COMMAND        FORMAT                        EXAMPLE                    Comment
--------------------------------------------------------------------------------
train          train (r0) (r1)...            train 0 2 45               Train records
load           load (r0) (r1) ...            load 0 51 2 3              Load records
clear          clear                         clear                      remove all records in  Recognizer
record         record / record (r0) (r1)...  record / record 0 79       Check record train status
vr             vr                            vr                         Check recognizer status
getsig         getsig (r)                    getsig 0                   Get signature of record (r)
sigtrain       sigtrain (r) (sig)            sigtrain 0 ZERO            Train one record(r) with signature(sig)
settings       settings                      settings                   Check current system settings
help           help                          help                       print this message
--------------------------------------------------------------------------------
sigtrain 0  Up
--------------------------------------------------------------------------------
Record:	0	Speak now
Record:	0	Speak again
Record:	0	Success
Success: 1
Record 0	Trained
SIG: Up
--------------------------------------------------------------------------------
sigtrain 1  Down
--------------------------------------------------------------------------------
Record:	1	Speak now
Record:	1	Speak again
Record:	1	Success
Success: 1
Record 1	Trained
SIG: Down
--------------------------------------------------------------------------------
sigtrain 2  Left
--------------------------------------------------------------------------------
Record:	2	Speak now
Record:	2	Speak again
Record:	2	Success
Success: 1
Record 2	Trained
SIG: Left
--------------------------------------------------------------------------------
load 0 1 2
--------------------------------------------------------------------------------
Load success: 0
Record 0	Record already in recognizer
Record 1	Record already in recognizer
Record 2	Record untrained
--------------------------------------------------------------------------------
record 0
--------------------------------------------------------------------------------
Check 1 records.
1 record trained.
0	-->	Trained
--------------------------------------------------------------------------------
record 1
--------------------------------------------------------------------------------
Check 1 records.
1 record trained.
1	-->	Trained
--------------------------------------------------------------------------------
record 2
--------------------------------------------------------------------------------
Check 1 records.
0 record trained.
2	-->	Untrained
--------------------------------------------------------------------------------
vr
--------------------------------------------------------------------------------
All voice records in recognizer: 0
Valid voice records in recognizer: 0
VR is not in group mode.
VR Index	Record		Comment
0		Unloaded	NONE
1		Unloaded	NONE
2		Unloaded	NONE
3		Unloaded	NONE
4		Unloaded	NONE
5		Unloaded	NONE
6		Unloaded	NONE
--------------------------------------------------------------------------------`

Core-drummer, 2019-10-11:
I solved my problem.  Instead 5V you can try connecting VCC to 3.3V pin. It helped me.

seifdaad, 2021-05-03:
remove the VCC, it will work

AgusOlivera, 2023-09-22:
Hey, same problem here. Does anyone know how to solve it? 
My module does not blink and even though it seems like the command has been trained, once I try to load it, I get the message "Record untrained".

I've followed the connections as the manual says. I don't get what the problem is. 

Please help!!!
[image]

AgusOlivera, 2023-10-06:
> > Hey, same problem here. Does anyone know how to solve it? My module does not blink and even though it seems like the command has been trained, once I try to load it, I get the message "Record untrained".
> > I've followed the connections as the manual says. I don't get what the problem is.
> > Please help!!! [image]
> 
> Did you find the solution?

No, I haven't. I'm absolutely clueless

AgusOlivera, 2023-10-06:
> Did you find the solution?

No, I didn't. I'm absolutely clueless

saad12-lab, 2023-10-06:
Nah bro . I am really stuck in it. Don't know how to solve it

Please help!!! [v3problem] <https://user-images.githubusercontent.com/65984351/270067586-ccf3a01d-9e9d-4f17-9c7c-8f2beb33d542.jpg>

saad12-lab, 2023-10-06:
I didn't find

saad12-lab, 2023-10-12:
Hy did you find the solution?

dakattack23, 2023-10-15:
I am having the same issue. Have you checked the "settings" command? Most of my settings are unknown. I've also tried using the vr sample bridge sketch to change the settings from the VR protocol codes but still no luck. 
[image]

saad12-lab, 2023-10-15:
Check your code

[image]<https://user-images.githubusercontent.com/148012541/275354445-9761c3ff-2949-4555-8aff-b4490ddaad20.png>

AgusOlivera, 2023-10-15:
> Check your code Get Outlook for iOS<https://aka.ms/o0ukef>
> […](#)

@saad12-lab But have you been able to solve the issue? Because I've checked everything I could, and still the module is not working. 

@dakattack23 I've also get the same response when I try the "settings" comand. And I've also tried to modify the parameters using the bridge example, but no change is made. It seems like the module is "read-only".

dakattack23, 2023-10-16:
@AgusOlivera agreed. I reached out to the manufacturer. i'll let you know if they provide any help.

Victor-R09, 2023-10-24:
> @AgusOlivera agreed. I reached out to the manufacturer. i'll let you know if they provide any help.

Any Luck?

dakattack23, 2023-10-24:
The manufacturer wasn't very helpful to find a solution, but they did offer to mail me a new one. Its on its way now. I'll keep you posted.

Rizky-9431, 2023-10-26:
Yeah I'm still waiting for the problem to be resolved :)

dakattack23, 2023-11-03:
The new module came and seems to be working fine. I was able to use the train sketch to record and load my commands.

StergiosT0, 2023-12-09:
Hello everyone, 
I bought a couple of NEW modules and nοne worked (train failed also Baud Rate in Settings output UNKNOWN). Luckily i found a old module and it worked fine.
I observe just a small difference between the new ones (already tried 5 new modules). See the attachments pictures below:
[image]
[image]
I already sent a email to elecouse support. I am waiting for any news..

khaled312001, 2024-01-09:
are you find the solution?

StergiosT0, 2024-01-09:
> are you find the solution?

Unfortunately not, i didn't get any answer from Elechouse yet..  I think something is wrong in updated version.
If anybody have any idea would be helpful..

Josualdo17, 2024-02-14:
any solution for this problem, I got the same problem with that

StergiosT0, 2024-02-14:
> any solution for this problem, I got the same problem with that

Personally unfortunately not, i also sent a email to Elechouse https://www.elechouse.com/product/speak-recognition-voice-recognition-module-v3/ support but they never replied.
Does anybody found a alternative solution?

robin8142, 2024-03-02:
https://www.youtube.com/watch?v=qSqbq6sm4rQ&t=304s buy this one, better support and more options. Elechouse never replied to mails.

wilson-elechouse, 2024-03-03:
Investigation Report on Voice Recognition V3 Product Issue

#### Executive Summary

This report addresses a critical issue identified in the Voice Recognition
V3 product line, which has affected approximately 3% of the units from the
production batch of August 2023. An estimated 200 units have been impacted,
causing inconvenience to a portion of our customers. It is important to
note that products purchased in 2024 are not affected by this issue.

#### Issue Description

The Voice Recognition V3 product has been found to exhibit 'untrained'
errors in about 3% of the units, specifically from the batch produced in
August 2023. This flaw has resulted in operational issues for some users,
significantly impacting their user experience.

#### Cause of the Issue

The root cause of the issue has been traced back to a change in the supply
chain. Our OEM factory partner initiated the use of a new supplier for the
W25Q16JVSSIQ chip, which has been identified as the source of the
'untrained' errors.

#### Resolution Measures

To address and rectify the issue, the following steps are being implemented
immediately:

1. **Chip Replacement:** The W25Q16JVSSIQ chip, identified as the cause of
the problem, will be replaced. This chip is available for purchase online,
facilitating a straightforward resolution process.

2. **Product Replacement for Customers:**
- **For buyers from elechouse.com:** Customers are encouraged to contact us
directly for a product exchange.
- **For buyers through other channels:** Customers have the option to
contact their seller for an exchange or reach out to elechouse.com to
purchase a replacement, with a special offer of buy one get one free.

#### Apology and Commitment

We acknowledge that the investigation into this issue took a significant
amount of time, and we deeply apologize for the inconvenience and
frustration it has caused our customers. We are committed to ensuring the
quality and reliability of our products and will continue to take all
necessary steps to prevent such issues from occurring in the future.

#### Conclusion

We are taking this matter very seriously and are committed to resolving it
promptly and efficiently. Our goal is to maintain trust and ensure the
satisfaction of our customers by providing quality products and responsive
customer service. We appreciate the patience and understanding of our
customers as we work to resolve this issue.

> Message ID: ***@***.***>
>

DARIOF8, 2024-06-24:
guys I have replaced the W25Q16JVSSIQ and it seems to work fine after 2weeks of testing literally everything .
for those of you who are capable of soldering smd componennts do your self a favor and replace the chip to save your time and money