elechouse/VoiceRecognitionV3 #17: Voice Commands work when powered by USB but not by wall adapter

bkeschinger, 2020-11-03:
Hello, I'm using the Elechouse VRv3 and have successfully trained it to work with the vr_sample_control_led program. I'm using this with an Elegoo Mega 2560 R3 Arduino board. When the board is plugged into my PC and receiving power from the USB, the program and commands work correctly! If I unplug the USB and power it via a 9V1A wall plug (USA), the program and voice commands no longer work.

Have you seen this before?

I put in a "digitalWrite(led, HIGH);" at the end of the Setup() function to ensure it's running and it is in both the USB and wall outlet situation. I'm unsure how to debug this from here! Thanks for your time!

nojlg, 2021-03-28:
I experienced the same problem and I think I have a solution. Voice recognition works fine when powered via my computer's USB port or a USB battery bank. When powered via an AC power source it doesn't work well or at all. The problem is caused by AC hum and different power supplies transmit different amounts of noise but they all seem to cause problems. I tried shielding and that didn't help either. Finally, I tried a cheap PC microphone that I had in my junk box and it works perfectly!

taf2, 2022-01-16:
Could a few filter capacitors help?  Maybe the power supply voltage is fluctuating a bit?