elechouse/VoiceRecognitionV3 #19: Using ESP32

ilhmfahmi, 2021-03-18:
i can't use this module with esp32 

Clear recognizer failed or timeout.

how to fix that

taf2, 2022-01-10:
@ilhmfahmi i modified the code replacing the SoftwareSerial dependency with Serial2 (builtin on esp32 boards)

For me I did this using a tinypico using pin 21 and 32.  Maybe this helps you too  see: https://github.com/taf2/VoiceRecognitionV3

Deewansonic, 2023-02-26:
have you gotten it solved with an ESP32 board? I seriously am in need of it.

taf2, 2023-02-26:
@Deewansonic yes using the tinypico see: https://unexpectedmaker.com/tinypico

Deewansonic, 2023-02-26:
@taf2 I have actually gone into https://unexpectedmaker.com/tinypico and I discovered that the ESP32 you meant was for TinyPico only and that may means that the solution you have given works only for ESP32 TinyPico. The ESP32 board I am dealing with is the ESP32-WROOM- 38 PIN Development Board, and I could not make it work and even using HardwareSerial doesn't seem to solve it.

Deewansonic, 2023-02-26:
could there be any assistance using the ESP32-WROOM- 38 PIN Development Board?

or could the new Seeed Studio XIAO ESP32C3 (https://www.seeedstudio.com/Seeed-XIAO-ESP32C3-p-5431.html) solve it just like the ESP32TinyPico did?

taf2, 2023-02-26:
@Deewansonic yeah tinypico is just a version of an ESP32 board pretty sure the chipset is identical to your board just in "esp32 pico d4"... regardless If I recall correctly the changes I made to get the library working is related to the use of Serial2 see my commits here: https://github.com/taf2/VoiceRecognitionV3/commits/master

zaid-alrowaili, 2023-09-22:
I have a problem saving the audio 
Can you help me 
![Uploading 269867729...]