elechouse/VoiceRecognitionV3 #25: VoiceRecognitionV3 is forgetting record I just trained

ICantSolveThatSht, 2023-05-27:
I'm using Elechouse VoiceRecognitionV3 module and program written from official library. I'm really new with this things and I really don't know pretty much anything. I doesn't really make sense what happened, could it be related to bad soldering or connection?
[image]

gubon13, 2023-08-07:
I am having the exact same issue with a new Elechouse Voice Recognition Module v3.1 purchased off of Amazon.

Every time, training says it's successful but checking the record shows it as Untrained and loading is consequently unsuccessful.

Checking settings returns:
```
Baud rate: UNKONOWN
Output IO Mode: UNKONOWN
Pulse Width: 20ms
Auto Load: enable
Group control by external IO: UNKNOWN STATUS
```

If I run vr, it shows all 7 records unloaded. The board came with trained records 16 and 37. If I try to load one of them, load is unsuccessful with error message: "Record already in recognizer". But again, run vr and it shows no records loaded.

I've tried using two different Arduino UNO boards (OEM) and two different Arduino NANO boards (clones), all with the same result.

I am using the defaults for everything. Default pins - D2:RX, D3:TX - and default speeds - myVR.begin(9600) / Serial.begin(115200). I've tried reversing the RX and TX pins, with and without updating the code to reflect the change. I've tried different USB cables. I've tried USB ports directly on the laptop, through a USB 3.0 hub, through a USB 2.0 hub, and through a docking station. I've tried from 2x different Apple Silicon Macbook Airs running macOS 13.3.1(a). I've tried "resetting" the module based on information I saw elsewhere stating to short the RX and TX pins and then apply VCC.

It's the same result every single time. It's as if the VR module is read-only. I stupidly tossed the packaging before testing so I cannot return. (And I soldered header pins on to the UART ports so that's not viable anyhow.)

Any help is greatly appreciated.

AgusOlivera, 2023-09-25:
Hey, I am having the same issue.

Did you manage to solve it?

Thanks!

gubon13, 2023-09-25:
@AgusOlivera I was not able to get it working. I spent countless hours trying. If you are still in the return window, I highly suggest returning it.

StergiosT0, 2023-12-06:
Hello everyone!
Did someone found any solution? Is there any difference in version v3.1?

Any help would be appreciated!

khaled312001, 2024-01-09:
Did somebody find any solution?

Robinarduino, 2024-03-03:
### Onderzoeksrapport over productprobleem met stemherkenning V3

#### Samenvatting

Dit rapport behandelt een kritiek probleem dat is geïdentificeerd in de
productlijn Spraakherkenning V3 en dat ongeveer 3% van
de apparaten uit de serie heeft getroffen de productiebatch van augustus 2023. Naar schatting
zijn 200 eenheden getroffen, wat ongemak veroorzaakt voor een deel van onze
klanten. Het is belangrijk op te merken dat producten die in 2024 zijn gekocht,
niet door dit probleem worden getroffen.

#### Beschrijving van het probleem

Het Spraakherkenning V3-product blijkt 'ongetrainde'
fouten te vertonen bij ongeveer 3% van de eenheden, met name bij de batch geproduceerd
in augustus 2023. Deze fout heeft voor sommige gebruikers geresulteerd in operationele problemen
, met aanzienlijke gevolgen hun gebruikerservaring.

#### Oorzaak van het probleem

De hoofdoorzaak van het probleem is terug te voeren op een verandering in de
toeleveringsketen. Onze OEM-fabriekspartner heeft het initiatief genomen om een ​​nieuwe
leverancier in te schakelen voor de W25Q16JVSSIQ-chip, die is geïdentificeerd als de
bron van de 'ongetrainde' fouten.

#### Oplossingsmaatregelen

Om het probleem aan te pakken en te verhelpen, worden de volgende stappen
onmiddellijk geïmplementeerd:

1. **Chipvervanging:** De W25Q16JVSSIQ-chip, geïdentificeerd als de
oorzaak van het probleem, zal worden vervangen. Deze chip kan
online worden gekocht, waardoor een eenvoudig oplossingsproces wordt vergemakkelijkt.

2. **Productvervanging voor klanten:**
- **Voor kopers van elechouse.com:** Klanten worden aangemoedigd om
rechtstreeks contact met ons op te nemen voor een productuitwisseling.
- **Voor kopers via andere kanalen:** Klanten hebben de mogelijkheid om
contact op te nemen met hun verkoper voor een omruiling of contact op te nemen met [elechouse.com](http://elechouse.com/) om
een ​​vervanging te kopen, met een speciale aanbieding om er één te kopen en er één gratis te krijgen.

#### Excuses en toewijding

We erkennen dat het onderzoek naar deze kwestie
veel tijd heeft gekost, en we bieden onze diepe excuses aan voor het
ongemak en de frustratie die dit bij onze klanten heeft veroorzaakt. Wij doen er
alles aan om de kwaliteit en betrouwbaarheid van onze producten te waarborgen en
zullen alle noodzakelijke stappen blijven ondernemen om te voorkomen dat dergelijke problemen
zich in de toekomst voordoen.

#### Conclusie

We nemen deze kwestie zeer serieus en streven ernaar
deze snel en efficiënt op te lossen. Ons doel is om het vertrouwen te behouden
en de tevredenheid van onze klanten te garanderen door kwaliteitsproducten
en een responsieve klantenservice te leveren. Wij waarderen het geduld
en begrip voor onze klanten terwijl we eraan werken dit probleem op te lossen.

StergiosT0, 2024-03-19:
New module arrived and works well!

I got one direct from the official store
https://www.elechouse.com/product/speak-recognition-voice-recognition-module-v3/

NoBoDy-Adk, 2024-04-07:
@stergiosT0 i got the same thing but I get the same issue everytime I run the code.

mCxNeyor, 2024-08-05:
Hey any one with solution please update