elechouse/VoiceRecognitionV3 #29: I have a problem saving the audio

zaid-alrowaili, 2023-09-22:
[image]

`Elechouse Voice Recognition V3 Module "train" sample.

Usage:

COMMAND FORMAT EXAMPLE Comment

train train (r0) (r1)... train 0 2 45 Train records
load load (r0) (r1) ... load 0 51 2 3 Load records
clear clear clear remove all records in Recognizer
record record / record (r0) (r1)... record / record 0 79 Check record train status
vr vr vr Check recognizer status
getsig getsig (r) getsig 0 Get signature of record (r)
sigtrain sigtrain (r) (sig) sigtrain 0 ZERO Train one record(r) with signature(sig)
settings settings settings Check current system settings
help help help print this message

sigtrain 0 Up

Record: 0 Speak now
Record: 0 Speak again
Record: 0 Success
Success: 1
Record 0 Trained
SIG: Up

sigtrain 1 Down

Record: 1 Speak now
Record: 1 Speak again
Record: 1 Success
Success: 1
Record 1 Trained
SIG: Down

sigtrain 2 Left

Record: 2 Speak now
Record: 2 Speak again
Record: 2 Success
Success: 1
Record 2 Trained
SIG: Left

load 0 1 2

Load success: 0
Record 0 Record already in recognizer
Record 1 Record already in recognizer
Record 2 Record untrained

record 0

Check 1 records.
1 record trained.
0 --> Trained

record 1

Check 1 records.
1 record trained.
1 --> Trained

record 2

Check 1 records.
0 record trained.
2 --> Untrained

vr

All voice records in recognizer: 0
Valid voice records in recognizer: 0
VR is not in group mode.
VR Index Record Comment
0 Unloaded NONE
1 Unloaded NONE
2 Unloaded NONE
3 Unloaded NONE
4 Unloaded NONE
5 Unloaded NONE
6 Unloaded NONE

AgusOlivera, 2023-09-25:
Hey, I'm having the exact same issue. 

Did you find any solution? Thanks

StergiosT0, 2023-12-07:
Did somebody find any solution?

khaled312001, 2024-01-09:
Did somebody find any solution?

StergiosT0, 2024-01-10:
> Did somebody find any solution?

Unfortunately not, i didn't get any answer from Elechouse yet.. I think something is wrong in updated version.
If anybody have any idea would be helpful..

paisp, 2024-02-02:
Hello SergiosT0. What is the difference between two components. Values? Tank you

robin8142, 2024-02-11:
did somebody get it worked? the new version. got the same problem here. very frustrating

StergiosT0, 2024-02-12:
> Hello SergiosT0. What is the difference between two components. Values? Tank you

I don't know exactly.. Just notice a optical difference, maybe is something like capacitor

StergiosT0, 2024-02-12:
> did somebody get it worked? the new version. got the same problem here. very frustrating

Personally unfortunately not, i also sent a email to Elechouse https://www.elechouse.com/product/speak-recognition-voice-recognition-module-v3/ support but they never replied.

StergiosT0, 2024-03-19:
New module arrived and works well!

I got one direct from the official store
https://www.elechouse.com/product/speak-recognition-voice-recognition-module-v3/