elechouse/VoiceRecognitionV3 #32: Voice Recognition Module V3 not recognized with Arduino Uno R4.

KnowledgeSeekerOnGitHub, 2024-03-10:
When using sample_train i get the output :

`train--------------------------------------------------------------------------------
Command Format Error!
--------------------------------------------------------------------------------
clear
--------------------------------------------------------------------------------
Clear recognizer failed or timeout.
--------------------------------------------------------------------------------
`

When using sample_control_led for simplicity the connection si not recognised.

Output:
`Elechouse Voice Recognition V3 Module
Control LED sample
Not find VoiceRecognitionModule.
Please check connection and restart Arduino.
`

it is failing the 
`
  if(myVR.clear() == 0){
    Serial.println("Recognizer cleared.");
  }else{
    Serial.println("Not find VoiceRecognitionModule.");
    Serial.println("Please check connection and restart Arduino.");
    while(1);
  }
`
check

`myVR.clear() == 0` is not being fulfilled

please help

KnowledgeSeekerOnGitHub, 2024-03-10:
Please also send a list of functions and uses, the doc provided do not have them and it is very confusing

wilson-elechouse, 2024-03-10:
do not use Arduino. Try to send command from PC directly

KnowledgeSeekerOnGitHub, 2024-03-10:
How? the module linked in the manua, is not on the elechouse page

KnowledgeSeekerOnGitHub, 2024-03-10:
> […](#)

even if i do not use arduino, this is arduino compatible, it is what it says, i need to use it in conjuction with an arduino for a project, even if i train it from PC, it need to work with my arduino for my purposes

KnowledgeSeekerOnGitHub, 2024-03-10:
> Please also send a list of functions and uses, the doc provided do not have them and it is very confusing

Is this possible?

wilson-elechouse, 2024-03-10:
https://www.elechouse.com/product/speak-recognition-voice-recognition-module-v3/
visit this page and you could get the manual and video guide

KnowledgeSeekerOnGitHub, 2024-03-10:
when visiting the product page im getting a 4040 not found

KnowledgeSeekerOnGitHub, 2024-03-10:
> > […](#)
> 
> even if i do not use arduino, this is arduino compatible, it is what it says, i need to use it in conjuction with an arduino for a project, even if i train it from PC, it need to work with my arduino for my purposes

@wilson-elechouse  can i also please get an answer to this?

wilson-elechouse, 2024-03-10:
I understand.
But any Arduino code problem might also cause the problem.
By sending commands to the module directly, we could check if the module
has problems

wilson-elechouse, 2024-03-10:
It might be your network problem. We could visit that page

KnowledgeSeekerOnGitHub, 2024-03-10:
> I understand. But any Arduino code problem might also cause the problem. By sending commands to the module directly, we could check if the module has problems KnowledgeSeeker ***@***.***> 于2024年3月10日周日 18:14写道：
> […](#)

the module does not have any problems, i have tried it in an Arduino Uno and an Arduino Mega, I can train, clear, load and use all commands
and the sample_led_control works

KnowledgeSeekerOnGitHub, 2024-03-10:
> […](#)

[image]

KnowledgeSeekerOnGitHub, 2024-03-10:
In my Arduino Uno Wifi R4, it is failing the 
```
if(myVR.clear() == 0){ Serial.println("Recognizer cleared."); }else{ Serial.println("Not find VoiceRecognitionModule."); Serial.println("Please check connection and restart Arduino."); while(1); }
check

myVR.clear() == 0 is not being fulfilled
```
What exactly does that do?

wilson-elechouse, 2024-03-10:
So it works with other Arduino version, but not on Arduino Uno Wifi R4?
manual： https://www.elechouse.com/elechouse/images/product/VR3/VR3_manual.pdf
video guide:
https://vimeo.com/94925137
https://vimeo.com/94926818

KnowledgeSeekerOnGitHub, 2024-03-10:
> So it works with other Arduino version, but not on Arduino Uno Wifi R4? manual： https://www.elechouse.com/elechouse/images/product/VR3/VR3_manual.pdf video guide: https://vimeo.com/94925137 https://vimeo.com/94926818 KnowledgeSeeker ***@***.***> 于2024年3月10日周日 18:23写道：
> […](#)

yes

KnowledgeSeekerOnGitHub, 2024-03-10:
@wilson-elechouse  are you a part of the official arduino discord?

KnowledgeSeekerOnGitHub, 2024-03-10:
> So it works with other Arduino version, but not on Arduino Uno Wifi R4? manual： https://www.elechouse.com/elechouse/images/product/VR3/VR3_manual.pdf video guide: https://vimeo.com/94925137 https://vimeo.com/94926818 KnowledgeSeeker ***@***.***> 于2024年3月10日周日 18:23写道：
> […](#)

i have seen all 3 of them.
when following the manual it works for the mega and the uno non-wifi. Just doesnt for the Wifi Rev4.

i have given the errors im getting on the post's initial msg

wilson-elechouse, 2024-03-10:
This library was created 8 years ago.
it uses softserial library. I suggest you check if Arduino wifi already
occupied those softserial pins for other purposes.

KnowledgeSeekerOnGitHub, 2024-03-10:
> This library was created 8 years ago. it uses softserial library. I suggest you check if Arduino wifi already occupied those softserial pins for other purposes. KnowledgeSeeker ***@***.***> 于2024年3月10日周日 18:28写道：
> […](#)

i checked the pinouts of the Wifi, it has the same pinouts of the Uno non-wifi, and it works in the non wifi.

KnowledgeSeekerOnGitHub, 2024-03-10:
> This library was created 8 years ago. it uses softserial library. I suggest you check if Arduino wifi already occupied those softserial pins for other purposes. KnowledgeSeeker ***@***.***> 于2024年3月10日周日 18:28写道：
> […](#)

Would the problems be because ESP8266 is integrated into the board?

KnowledgeSeekerOnGitHub, 2024-03-10:
Please reply

KnowledgeSeekerOnGitHub, 2024-03-10:
can i get a list of what the functions do?

KnowledgeSeekerOnGitHub, 2024-03-10:
```
17:06:35.764 -> Elechouse Voice Recognition V3 Module
17:06:35.764 -> Control LED sample
17:06:36.734 -> Not find VoiceRecognitionModule.
17:06:36.734 -> Please check connection and restart Arduino.
```
what could lead to this error in sample_led_control

wilson-elechouse, 2024-03-10:
UNO R4 WiFi has different MCU with UNO R3
I suggest you study the softserial library and change the code
https://docs.arduino.cc/learn/built-in-libraries/software-serial/

KnowledgeSeekerOnGitHub, 2024-03-10:
Thanks a lot i'll look into it right away. Anything else you can suggest?

KnowledgeSeekerOnGitHub, 2024-03-10:
Also i know im asking this a lot but is there a list of functions available for study so i can easily edit the code?

wilson-elechouse, 2024-03-10:
https://dropmeafile.com/#978386e3d3

KnowledgeSeekerOnGitHub, 2024-03-10:
I have that, i was asking for the description of all methods and functions present in the library, like
```
myVR.clear()
```
what exactly does that do?

wilson-elechouse, 2024-03-10:
Sorry, I am not a software developer.
This command should be the function of deleting all commands in the recognizer.

KnowledgeSeekerOnGitHub, 2024-03-10:
> […](#)

Do you have the contact or any way i can contact the person who has developed this library please?

KnowledgeSeekerOnGH, 2024-03-11:
Hello?

KnowledgeSeekerOnGitHub, 2024-03-11:
What exactly is `` VR myVR(2,3)``` that is present in each sample code?

KnowledgeSeekerOnGitHub, 2024-03-11:
Hello?

wilson-elechouse, 2024-03-12:
Sorry for replying late
that's the function to define the softserial pins